ROBOTIS-GIT/OpenCR #293: Error downloading

yxldd, 2022-01-19:
When  I  tried to add OpenCR manager on arduino IDE(on windows),  it always tell me that 
"Error downloading https://raw.githubusercontent.com/ROBOTIS-GIT/OpenCR/master/arduino/opencr_release/package_opencr_index.json"

ROBOTIS-Will, 2022-01-19:
Hi @yxldd 
Please check if you can access the GitHub in your region.
Try opening the json file location on your web browser and see if you can read the content.
https://raw.githubusercontent.com/ROBOTIS-GIT/OpenCR/master/arduino/opencr_release/package_opencr_index.json
Thanks.

yxldd, 2022-01-19:
> Hi @yxldd Please check if you can access the GitHub in your region. Try opening the json file location on your web browser and see if you can read the content. https://raw.githubusercontent.com/ROBOTIS-GIT/OpenCR/master/arduino/opencr_release/package_opencr_index.json Thanks.

Thanks for reply. I can access GitHub in my region, and post this issue successfully. And I can also see the content about the .json in my web browser.

yxldd, 2022-01-19:
[image]
when I click the tools->Boards Manager, the arduino IDE will tell me this.
I am a new university student ,and maybe dont have enough ability to solve it.
THANKS FOR HELPING.

closed 2022-01-21 by yxldd