ROBOTIS-GIT/OpenCR #296: CAN communication OpenCR

kaif-46, 2022-01-31:
How can I adjust the frequency of CAN communication in OpenCR1.0 using the Arduino library?

ROBOTIS-Will, 2022-02-03:
@kaif-46 
Please refer to the example in OpenCR.
https://github.com/ROBOTIS-GIT/OpenCR/blob/master/arduino/opencr_arduino/opencr/libraries/OpenCR/examples/10.%20Etc/CAN/canTxRxByte/canTxRxByte.ino#L64

Supported Baudrates are
```
 * BAUDRATE :
 *   CAN_BAUD_125K
 *   CAN_BAUD_250K
 *   CAN_BAUD_500K
 *   CAN_BAUD_1000K
```

kaif-46, 2022-02-03:
@ROBOTIS-Will 
I have taken a look at it but along with the baud rater I even need to set the frequency of communication as my motor supports 8MHZ communication only.
I have done the same thing using Arduino by using an external CAN Module(MCP2515) and the mcp2515.h or mcp2515_can.h libraries.

ROBOTIS-Will, 2022-02-03:
@kaif-46 
STM32F7 and MCP2515 both support up to 1Mbps CAN bus.
Thanks.

kaif-46, 2022-02-03:
@ROBOTIS-Will 
Thank you for the information but I need to know how can I bring down the frequency to 8MHz by using the CAN.h library used in the examples.