ROBOTIS-GIT/OpenCR #297: how to model Turtlebot3_Segway ?

swgu931, 2022-02-09:
Hello!

I appreciate your valuable works especially turtlebot3 and segway.
I wanted to find the turtlebot3 segway modeling before designing controller but I cannot find it inside of the source code below.
https://github.com/ROBOTIS-GIT/OpenCR/tree/master/arduino/opencr_arduino/opencr/libraries/turtlebot3/examples/turtlebot3_friends/turtlebot3_segway

I am doubtful that you control the turtlebot3_segway without any modeling but heuristic or "trial and error" approach.

Thanks in advance for your help.

ROBOTIS-Will, 2022-02-10:
Hi @swgu931 

Thank you for your interest in TurtleBot3 Friends!
You can find the modeling of the segway from TurtleBot3 eManual below.
https://emanual.robotis.com/docs/en/platform/turtlebot3/locomotion/#turtlebot3-friends-segway
Thank you!

swgu931, 2022-02-11:
I mean the model that you control with PID controller, namely the state space equations or funtional equation for control engineers.
The following could be one of examples. 
x_dot = Ax + Bu
y = Cx

Thanks.

ROBOTIS-Will, 2022-02-15:
Hi @swgu931 

[The PID controller model of the example](https://github.com/ROBOTIS-GIT/OpenCR/blob/master/arduino/opencr_arduino/opencr/libraries/turtlebot3/examples/turtlebot3_friends/turtlebot3_segway/turtlebot3_segway.ino#L178-L194) is tuned heuristically based on the IMU pitch reading of the TurtleBot3 Segway.
Thank you.

swgu931, 2022-02-28:
I made the similar one that you tested on the youtube. you can see
I have two questions below.

1) what is the right one about x, y direction of IMU on the OpenCR board ?
2) The value of P, I, D in the source code is your final tuned value, isn't it ?

[image]

Thanks in advance.