ROBOTIS-GIT/OpenCR #299: I have trouble installing OpenCR package into Arduino IDE

hirasawaryo, 2022-03-22:
I failed to install the OpenCR package (1.4.19) via boards manager of Arduino IDE. I tried to install by three computers, however I failed all computers. I tried to do by using some networks, but I couldn't. I can install some other board packages, but I cannot install OpenCR package.

Please help me.

Manual I read: 
https://emanual.robotis.com/docs/en/parts/controller/opencr10/#install-on-windows

Configuration : 
PC-1 : Windows 10 / Arduino IDE 1.8.19
PC-2 : Windows 10 / Arduino IDE 1.8.19
PC-3 : Ubuntu 18.04 / Arduino IDE 1.8.19

Procedure : 
1. Download Arduino IDE
2. File -> Preferencies -> Additional Boards Manager URLs : https://raw.githubusercontent.com/ROBOTIS-GIT/OpenCR/master/arduino/opencr_release/package_opencr_index.json
3. Tools -> Board -> Board Manager -> OpenCR -> Install

Error massage (Board Manager window):
Error downloading https://github.com/ROBOTIS-GIT/OpenCR/releases/download/1.4.19/opencr.tar.bz2

Error massage (command line):
Error downloading https://github.com/ROBOTIS-GIT/OpenCR/releases/download/1.4.19/opencr.tar.bz2
java.lang.RuntimeException: java.lang.Exception: Error downloading https://github.com/ROBOTIS-GIT/OpenCR/releases/download/1.4.19/opencr.tar.bz2
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$2(ContributionManagerUI.java:179)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.Exception: Error downloading https://github.com/ROBOTIS-GIT/OpenCR/releases/download/1.4.19/opencr.tar.bz2
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:145)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:79)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:56)
	at cc.arduino.contributions.packages.ContributionInstaller.install(ContributionInstaller.java:100)
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$2(ContributionManagerUI.java:172)
	... 1 more
Caused by: javax.net.ssl.SSLHandshakeException: sun.security.validator.ValidatorException: PKIX path building failed: sun.security.provider.certpath.SunCertPathBuilderException: unable to find valid certification path to requested target
	at sun.security.ssl.Alerts.getSSLException(Alerts.java:192)
	at sun.security.ssl.SSLSocketImpl.fatal(SSLSocketImpl.java:1946)
	at sun.security.ssl.Handshaker.fatalSE(Handshaker.java:316)
	at sun.security.ssl.Handshaker.fatalSE(Handshaker.java:310)
	at sun.security.ssl.ClientHandshaker.serverCertificate(ClientHandshaker.java:1639)
	at sun.security.ssl.ClientHandshaker.processMessage(ClientHandshaker.java:223)
	at sun.security.ssl.Handshaker.processLoop(Handshaker.java:1037)
	at sun.security.ssl.Handshaker.process_record(Handshaker.java:965)
	at sun.security.ssl.SSLSocketImpl.readRecord(SSLSocketImpl.java:1064)
	at sun.security.ssl.SSLSocketImpl.performInitialHandshake(SSLSocketImpl.java:1367)
	at sun.security.ssl.SSLSocketImpl.startHandshake(SSLSocketImpl.java:1395)
	at sun.security.ssl.SSLSocketImpl.startHandshake(SSLSocketImpl.java:1379)
	at sun.net.www.protocol.https.HttpsClient.afterConnect(HttpsClient.java:559)
	at sun.net.www.protocol.https.AbstractDelegateHttpsURLConnection.connect(AbstractDelegateHttpsURLConnection.java:185)
	at sun.net.www.protocol.http.HttpURLConnection.followRedirect0(HttpURLConnection.java:2729)
	at sun.net.www.protocol.http.HttpURLConnection.followRedirect(HttpURLConnection.java:2641)
	at sun.net.www.protocol.http.HttpURLConnection.getInputStream0(HttpURLConnection.java:1824)
	at sun.net.www.protocol.http.HttpURLConnection.getInputStream(HttpURLConnection.java:1492)
	at java.net.HttpURLConnection.getResponseCode(HttpURLConnection.java:480)
	at sun.net.www.protocol.https.HttpsURLConnectionImpl.getResponseCode(HttpsURLConnectionImpl.java:347)
	at cc.arduino.utils.network.HttpConnectionManager.makeConnection(HttpConnectionManager.java:139)
	at cc.arduino.utils.network.HttpConnectionManager.makeConnection(HttpConnectionManager.java:96)
	at cc.arduino.utils.network.FileDownloader.openConnectionAndFillTheFile(FileDownloader.java:217)
	at cc.arduino.utils.network.FileDownloader.downloadFile(FileDownloader.java:176)
	at cc.arduino.utils.network.FileDownloader.download(FileDownloader.java:125)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:143)
	... 5 more
Caused by: sun.security.validator.ValidatorException: PKIX path building failed: sun.security.provider.certpath.SunCertPathBuilderException: unable to find valid certification path to requested target
	at sun.security.validator.PKIXValidator.doBuild(PKIXValidator.java:397)
	at sun.security.validator.PKIXValidator.engineValidate(PKIXValidator.java:302)
	at sun.security.validator.Validator.validate(Validator.java:262)
	at sun.security.ssl.X509TrustManagerImpl.validate(X509TrustManagerImpl.java:324)
	at sun.security.ssl.X509TrustManagerImpl.checkTrusted(X509TrustManagerImpl.java:229)
	at sun.security.ssl.X509TrustManagerImpl.checkServerTrusted(X509TrustManagerImpl.java:124)
	at sun.security.ssl.ClientHandshaker.serverCertificate(ClientHandshaker.java:1621)
	... 26 more
Caused by: sun.security.provider.certpath.SunCertPathBuilderException: unable to find valid certification path to requested target
	at sun.security.provider.certpath.SunCertPathBuilder.build(SunCertPathBuilder.java:141)
	at sun.security.provider.certpath.SunCertPathBuilder.engineBuild(SunCertPathBuilder.java:126)
	at java.security.cert.CertPathBuilder.build(CertPathBuilder.java:280)
	at sun.security.validator.PKIXValidator.doBuild(PKIXValidator.java:392)
	... 32 more

ROBOTIS-Shibata, 2022-04-04:
Hi, @hirasawaryo 
Did you start the OpenCR boot loader?
Before you can upload the program to OpenCR, you need to operate the switch on OpenCR to start the Boot loader.
See the link below for details.

https://emanual.robotis.com/docs/en/parts/controller/opencr10/#bootloader

(日本語版)https://emanual.robotis.com/docs/en/parts/controller/opencr10_jp/#%E3%83%96%E3%83%BC%E3%83%88%E3%83%AD%E3%83%BC%E3%83%80%E3%83%BC

ROBOTIS-Will, 2022-04-21:
@hirasawaryo 
Can you directly download the board manager from the link below?

https://github.com/ROBOTIS-GIT/OpenCR/releases/download/1.4.19/opencr.tar.bz2

Sometimes Arduino IDE returns error while installing the board manager, and usually retrying the board manager installation resolves the issue.

Did you install the Arduino IDE via Windows app?

sunghowoo, 2025-02-24:
This issue has been inactive for a long time, we will proceed with closing it. If you have any further concerns, please feel free to reopen the issue or create a new one.

closed 2025-02-24 by sunghowoo