ROBOTIS-GIT/OpenCR #301: turtlebot3_ros2 cant create publisher at bringup

FranciscoBarCa, 2022-04-02:
Hello, 
I'm trying to  compile and upload the OpenCR firmware for my turtlebot3 running ROS2 foxy, the upload is successful and the button test to check the connection to the motors runs smoothly. The issue arises while doing the bringup of the SBC using the command 
` ros2 launch turtlebot3_bringup robot.launch.py`
It gives out an error during the battery publisher creation, funnily enough, this error does not occur while uploading the firmware using the precompiled firmware at https://emanual.robotis.com/docs/en/platform/turtlebot3/opencr_setup/#opencr-setup .
Console log:
```
ubuntu@ubuntu:~$  ros2 launch turtlebot3_bringup robot.launch.py
[INFO] [launch]: All log files can be found below /home/ubuntu/.ros/log/2022-04-02-15-49-04-907769-ubuntu-2019
[INFO] [launch]: Default logging verbosity is set to INFO
urdf_file_name : turtlebot3_burger.urdf
/opt/ros/foxy/share/hls_lfcd_lds_driver/launch/hlds_laser.launch.py:46: UserWarning: The parameter 'node_executable' is deprecated, use 'executable' instead
  Node(
/opt/ros/foxy/share/hls_lfcd_lds_driver/launch/hlds_laser.launch.py:46: UserWarning: The parameter 'node_name' is deprecated, use 'name' instead
  Node(
[INFO] [robot_state_publisher-1]: process started with pid [2021]
[INFO] [hlds_laser_publisher-2]: process started with pid [2023]
[INFO] [turtlebot3_ros-3]: process started with pid [2025]
[hlds_laser_publisher-2] [INFO] [1648914547.749399343] [hlds_laser_publisher]: Init hlds_laser_publisher Node Main
[hlds_laser_publisher-2] [INFO] [1648914547.750329708] [hlds_laser_publisher]: port : /dev/ttyUSB0 frame_id : base_scan
[turtlebot3_ros-3] [INFO] [1648914547.791459708] [turtlebot3_node]: Init TurtleBot3 Node Main
[turtlebot3_ros-3] [INFO] [1648914547.796492260] [turtlebot3_node]: Init DynamixelSDKWrapper
[turtlebot3_ros-3] [INFO] [1648914547.802845853] [DynamixelSDKWrapper]: Succeeded to open the port(/dev/ttyACM0)!
[turtlebot3_ros-3] [INFO] [1648914547.811697728] [DynamixelSDKWrapper]: Succeeded to change the baudrate!
[robot_state_publisher-1] Parsing robot urdf xml string.
[robot_state_publisher-1] Link base_link had 5 children
[robot_state_publisher-1] Link caster_back_link had 0 children
[robot_state_publisher-1] Link imu_link had 0 children
[robot_state_publisher-1] Link base_scan had 0 children
[robot_state_publisher-1] Link wheel_left_link had 0 children
[robot_state_publisher-1] Link wheel_right_link had 0 children
[robot_state_publisher-1] [INFO] [1648914547.833617572] [robot_state_publisher]: got segment base_footprint
[robot_state_publisher-1] [INFO] [1648914547.834385645] [robot_state_publisher]: got segment base_link
[robot_state_publisher-1] [INFO] [1648914547.834537676] [robot_state_publisher]: got segment base_scan
[robot_state_publisher-1] [INFO] [1648914547.834639551] [robot_state_publisher]: got segment caster_back_link
[robot_state_publisher-1] [INFO] [1648914547.834733406] [robot_state_publisher]: got segment imu_link
[robot_state_publisher-1] [INFO] [1648914547.834825333] [robot_state_publisher]: got segment wheel_left_link
[robot_state_publisher-1] [INFO] [1648914547.834966426] [robot_state_publisher]: got segment wheel_right_link
[turtlebot3_ros-3] [ERROR] [1648914547.847055489] [turtlebot3_node]: Failed connection with Devices
[turtlebot3_ros-3] [INFO] [1648914547.862535333] [turtlebot3_node]: Add Motors
[turtlebot3_ros-3] [INFO] [1648914547.863344135] [turtlebot3_node]: Add Wheels
[turtlebot3_ros-3] [INFO] [1648914547.863910593] [turtlebot3_node]: Add Sensors
[turtlebot3_ros-3] terminate called after throwing an instance of 'rclcpp::exceptions::RCLError'
[turtlebot3_ros-3]   what():  could not create publisher: rcl node's context is invalid, at /tmp/binarydeb/ros-foxy-rcl-1.1.11/src/rcl/node.c:441
[ERROR] [turtlebot3_ros-3]: process has died [pid 2025, exit code -6, cmd '/home/ubuntu/turtlebot3_ws/install/turtlebot3_node/lib/turtlebot3_node/turtlebot3_ros -i /dev/ttyACM0 --ros-args --params-file /home/ubuntu/turtlebot3_ws/install/turtlebot3_bringup/share/turtlebot3_bringup/param/burger.yaml'].
```
I would really appreciate any help on the subject.
Kind regards.

FranciscoBarCa, 2022-04-09:
Hi, 
I am might have used the wrong word as firmware probably isn't the most appropriate in this case.
Regardless, I am using the latest release of the OpenCR board at [https://raw.githubusercontent.com/ROBOTIS-GIT/OpenCR/master/arduino/opencr_release/package_opencr_index.json](https://raw.githubusercontent.com/ROBOTIS-GIT/OpenCR/master/arduino/opencr_release/package_opencr_index.json), the version 1.4.19.
As for the sketch im using the example sketch at  Turtlebot3ROS2/turtlebot_burger.
Hope it helps.

rzeeshan565, 2022-04-09:
Which SBC,
Because I guess arduino based OpenCR firmware doesn't work.on Raspberry pi.

FranciscoBarCa, 2022-04-09:
No, obviously im uploading it to the OpenCR.
But the SBC i'm running ROS2 is as you are suggesting a raspberry pi 3b. I talked about the SBC because i notice the failure while doing the bringup of ros. I'm not very knowledgeable at the matter, but the error seems to suggest the SBC doesn't receive data from the OpenCR.

FranciscoBarCa, 2022-04-10:
Hello, this morning I've revisited the issue as I had left it on standby.
I've managed to solve it by downgrading the DYNAMIXEL2Arduino library to `v0.3.0` as suggested in the similar issue #245.
Later versions might also work but i haven't had the need to try another.
I will now close the issue.
Thanks for the help.

closed 2022-04-10 by FranciscoBarCa

rzeeshan565, 2022-04-10:
Okay that is great.
Can you please share the steps to downgrade the library?

rzeeshan565, 2022-04-10:
thank you @FranciscoBarCa, that resolved my problem.