ROBOTIS-GIT/OpenCR #304: Connecting a Servo to the OpenCR

pitosalas, 2022-04-08:
Looking around the source code and googling around, it seems to be that the Turtlebot3 / OpenCR board is set up to accept a servo connector. I see there's a package called servo etc. My question is where on the OpenCR board should we attach the servo connector? THank you!

ROBOTIS-Shibata, 2022-04-09:
Hi,
If it is meaning PWM servo, ROBOTIS e-manual has the Pin assign description.
https://emanual.robotis.com/docs/en/parts/controller/opencr10/

pitosalas, 2022-04-09:
I see. Can you say whether how to enable the topic via rosserial on the OpenCR or does that happen "automatically"?

pitosalas, 2022-04-12:
> Hi,
> This OpenCR ROS examples may help you to implement the publisher / subscriber for PWM motor control.
> Similar sub/pubs are implemented in the TurtleBot3 ROS example.

Thank you. I cannot locate the OpenCR ROS examples. And the TB3 ROS example shows the ROS side. I am asking about the OpenCR side. For example:

`self.bumper_sub = rospy.Subscriber('sensor_state', SensorState, self.get_bumper, queue_size = 1)`

I need the code on the OpenCR side that would publish sensor_state

Any pointers?

Pito Salas
Faculty, Computer Science
Brandeis University

ROBOTIS-Will, 2022-04-21:
@pitosalas 
I'm sorry about the delayed response.
ROS examples in OpenCR board manager have those pub/sub implmentations.
For example, the bumper status is published via "bumper" topic in the example below.
https://github.com/ROBOTIS-GIT/OpenCR/blob/master/arduino/opencr_arduino/opencr/libraries/ROS/examples/02.%20Sensors/a_bumper/a_bumper.ino#L58
Thank you.

pitosalas, 2022-04-27:
Thanks… Is there a developer guide for OpenCR? Because I understand the code you linked me to but I dont know where the “main loop” is for opencr where I would add rosserial code to subscribe to a topic and send the right commands to the servo. 

Pito Salas
Faculty, Computer Science
Brandeis University

yun-goon, 2025-08-05:
Hello, and sorry for the late reply.

We believe that the examples should have been sufficient for you to resolve the issue. If you are still running into problems while writing your code, please share what you have so far. It would be much easier for us to help you troubleshoot.

Since this issue is very old, we're going to close it. If you have any new questions or run into another problem, please feel free to open a new issue.

Thank you for your understanding!

closed 2025-08-05 by yun-goon