ROBOTIS-GIT/OpenCR #307: OpenCR1.0 with TPS10 using OLLO Library

huybeme, 2022-05-09:
Hello, I have been trying to use the TPS10 temperature sensor with the OpenCR1.0 using the OpenCR 5 pin ports. I have noticed that using it with the arduino IDE, I usually get outputs between -20 to -18.

Looking into the OLLO library, I modified its code a bit and seem to find that the vvalue seems like the correct output for temperature.

Could I please confirm this since I am now getting values that I except to see.

ROBOTIS-Will, 2022-06-16:
Hi @huybeme 
I'm sorry about the delayed response.
We'll check with this and get back to you as soon as possible.

ROBOTIS-David, 2022-06-16:
Hi, @huybeme 

Thank for raising a issue ticket, and sorry for missing your ticket. 

I found a OLLO library has a range issue. The resolved version can be refered at following branch ([develop-ollo](https://github.com/ROBOTIS-GIT/OpenCR/tree/develop-ollo))

[image]
> Test Result 

To apply the fixed, please update the following files by referenceing [ollo.cpp](https://github.com/ROBOTIS-GIT/OpenCR/commit/f6b28ae6f24b5511db3ed869c1f6d6f8203cb74e#diff-d663178c5d830a08dbd62a01a66e72cdff46bf925d4ca2eaab8ae27496cf7f79) locally. 

The update will come later. 

Thank you

sunghowoo, 2025-02-24:
This issue has been inactive for a long time, we will proceed with closing it. If you have any further concerns, please feel free to reopen the issue or create a new one.

PR #312 (2022-06-17): Develop ollo
fixes #307
closed 2025-02-24 by sunghowoo