ROBOTIS-GIT/OpenCR #308: Using RS485 port on OpenCR to communicate with camera

cmtchow, 2022-05-10:
I'm new to this field and just started to learn about OpenCR board. 

I'm trying to use one of the RS485 ports on OpenCR to send and receive data from a  camera that uses RS485 communication. 
My question is, if it possible to use the RS485 port on OpenCR other than motors? If so, how can I begin communication and which pin am I supposed to use in pinMode as output and input? 

Thank you.

shiba-8ro, 2022-05-11:
@cmtchow 
OpenCR RS485 port only supports Dynamixel motors.
To be precise, the communication circuit complies with RS485 standard, but we do not provide library for communicating with other devices than Dynamixel motors. So if you connect RS485 Camera to OpenCR, you need to DIY the driver.
By the way, which model of RS485 Camera do you use?

cmtchow, 2022-05-12:
Thank you for your reply. 

Sorry to ask some novice questions. If I have to DIY the driver, where and how can I DIY it? 

For the camera I'm using,  I'm using Pepperl+FUCHS's PGV100-F200A-R4-V19 as a camera for camera-based track guidance. 
https://www.pepperl-fuchs.com/global/en/classid_3334.htm?view=productdetails&prodid=60355

Thank you.

shiba-8ro, 2022-05-12:
The camera does not appear to be intended for use on Emmbeded boards.
You should try it from PC first, using the USB-RS485 (M12 connector) interface recommended by the camera manufacturer.
And, in order to DIY the Arduino driver, you need literacy to parse the Arduino library.
[Dynamixel2Arduino](https://github.com/ROBOTIS-GIT/Dynamixel2Arduino/tree/master/src)
[Arduino Library : How to Create Your Own Arduino Library](https://youtu.be/z2BstX2ff_g)

ROBOTIS-Will, 2022-05-12:
Hi @cmtchow 

The current OpenCR library use the RS485 port only for DYNAMIXEL with DYNAMIXEL SDK library.
https://github.com/ROBOTIS-GIT/OpenCR/tree/master/arduino/opencr_arduino/opencr/libraries/DynamixelSDK/include/dynamixel_sdk

Since there isn't any driver(or Arduino library for OpenCR) for the camera module, you'll need to write your own send and receive/parsing data function.
However, this will be quite difficult if you don't have decent understanding about both camera and OpenCR.

As @shiba-8ro mentioned in the reply above, I'd recommend trying the camera on the PC first to understand how it works.

Thanks.

sunghowoo, 2025-02-24:
This issue has been inactive for a long time, I will proceed with closing it.

closed 2025-02-24 by sunghowoo