ROBOTIS-GIT/OpenCR #310: TB3/ROS2 OpenCR communication error

giovanidiniz, 2022-06-14:
Hi, guys, 

I'm updating my ROS stack to use Humble. I understand it is not fully supported yet, but was hoping to start making some progress towards that. I compile the firmware without issues using the Arduino IDE ( OpenCR version 1.4.19/DXL2Arduino 0.6.0)

When I run the bringup node, I get the following error:

```
[turtlebot3_ros-2] [ERROR] [1655217926.169649893] [turtlebot3_node]: Failed connection with Devices
[turtlebot3_ros-2] [INFO] [1655217926.170118367] [turtlebot3_node]: Add Motors
[turtlebot3_ros-2] [INFO] [1655217926.170151405] [turtlebot3_node]: Add Wheels
[turtlebot3_ros-2] [INFO] [1655217926.170163220] [turtlebot3_node]: Add Sensors
[turtlebot3_ros-2] terminate called after throwing an instance of 'rclcpp::exceptions::RCLError'
```

which points me to this check (turtlebot3.cpp L:77)
```
if (dxl_sdk_wrapper_->is_connected_to_device()) 
```
I also verified the board is connected properly and the DXL SDK instantiates properly:
```
[turtlebot3_ros-2] [INFO] [1655217926.133157842] [turtlebot3_node]: Init TurtleBot3 Node Main
[turtlebot3_ros-2] [INFO] [1655217926.133277332] [turtlebot3_node]: Init DynamixelSDKWrapper
[turtlebot3_ros-2] [INFO] [1655217926.133586270] [DynamixelSDKWrapper]: Succeeded to open the port(/dev/ttyACM0)!
[turtlebot3_ros-2] [INFO] [1655217926.135440763] [DynamixelSDKWrapper]: Succeeded to change the baudrate!
```

Hoping someone could help me out in this. I'm in a bind now and have custom firmware version mods for my own robots to work on but I'm stuck trying to get the vanilla TB3 FW running.

P.S: any chance this is related (https://github.com/ROBOTIS-GIT/turtlebot3/issues/744)

Appreciate the assist

giovanidiniz, 2022-06-16:
Hi, there.
I had success running with OpenCR 1.4.19 and downgrading Dynamixel2Arduino to 0.3.0, but I'll adjust accordingly now.
I am having issues with the OpenCR control table not being read properly by the ROS2 node, but I'll open a different issue to handle that. Your changes solved the problem for me

closed 2022-06-16 by giovanidiniz

ROBOTIS-Will, 2022-06-24:
@giovanidiniz 
Hi, the latest release resolves this issue with ROS2.
Please update the DYNAMIXEL2Arduino to `0.6.0` and OpenCR to `1.5.0`.
We'll release the binary files for uploading the OpenCR firmware from RPi.
Thanks!

giovanidiniz, 2022-06-25:
@ROBOTIS-Will Appreciate the heads up!

I'm up and running with my custom firmware on a 4-wheel robot.
Thanks so much