ROBOTIS-GIT/OpenCR #311: OpenCR 1.0 firmware upload is failed because of flash erase timeout

chjej202, 2022-06-15:
Hello, 

I got an error during uploading my firmware on OpenCR 1.0 board.

The error message is shown in the following screenshot.
[image]

I noticed that the error is happened during flash erase, and I found that the timeout is happened because flash erase time took more than 3 seconds.

https://github.com/ROBOTIS-GIT/OpenCR/blob/c37bc731ffb84693f5742dd692d36656fffef72f/arduino/opencr_develop/opencr_ld/opencr_ld.c#L725

For this reason, I changed the above code from 3000 to 5000 to increase the timeout value. 
After I build and replace opencr_ld command, it works fine for me.

I think 3 seconds is too short to upload large firmware, so please increase the timeout value.

ROBOTIS-Will, 2022-06-16:
Hi @chjej202 

Thank you for reporting the issue.
The timeout error during the flash upload is unusual as we test the upload procedure with provided firmware.
Could you give us more information about your development environment?
Assuming from the size of the 517KB binary file, are you uploading a customized firmware?

chjej202, 2022-06-16:
The timeout is happened when the binary file is very large. 
In my environment, it took around 3 seconds for uploading 500KB-size binary  file. (510KB - flash upload ok (flash erase time: 2.9xx seconds), 512KB - flash erase timeout happened)
I think that flash erase took several seconds to erase several hundreds of kilobytes.

Because of this issue, I cannot fully use all the flash memory provided by OpenCR 1.0 (768KB) unless the timeout value is not changed.

Our binary file includes lots of memory-burden examples. It includes arduino tensorflow lite for person detection, jpeg decoder, arduino cam, and code for accessing sensors and motors.

ROBOTIS-Will, 2022-06-16:
@chjej202 

Thanks for the details and how interesting that you are running the tensorflow on OpenCR!

Quite long ago, I uploaded the [OV7725 + TFT OpenCR example](https://github.com/ROBOTIS-GIT/OpenCR/tree/master/arduino/opencr_arduino/opencr/libraries/OpenCR/examples/10.%20Etc/CAMERA/ov7725_al422b), but wasn't able to expand to the proper image processing or machine learning.

If your project can be disclosed to the public via OpenCR eManual, it'd be an amazing OpenCR tensorflow example.

chjej202, 2022-06-16:
I uploaded my code related to use Arduino tensorflow lite on OpenCR 1.0.

I only changed and tested a person detection example only, but I think you can easily modify and run other examples also.
You can check the commit logs to find which parts are modified.

Modified code to run ArduCAM library on OpenCR 1.0 board: https://github.com/chjej202/Arduino
Modified code to run Arduino Tensorflow lite library on OpenCR 1.0 board: https://github.com/chjej202/Arduino_TensorFlowLite

Arduino tensorflow lite code was modified based on the code uploaded in https://www.arduino.cc/reference/en/libraries/arduino_tensorflowlite

ROBOTIS-Will, 2022-06-17:
@chjej202 
Super!!!
Thanks for sharing your repos.
We'll try them and share with developers through eManual and OpenCR examples.
Have a great weekend ahead.

sunghowoo, 2025-06-20:
Since this issue seems to be resolved, we will go ahead and close it.

closed 2025-06-20 by sunghowoo