ROBOTIS-GIT/OpenCR #317: Offline time sync help

fabiocyro, 2022-09-23:
Hi, I have a turtlebot3 and I would like to run the SLAM without having to be connected to the internet. But the raspberry pi does not get the current time if it is not connected to internet as everyone knows.

The thing is that I have a RTC module that can be attached to the OpenCR, and my question is:

Can I connect this module to the OpenCR in order to update the time for the rasperry? 

I have no idea if this is possible, but if yes, can someone help me?

johnauld, 2022-10-21:
The OpenCR has its own RTC built into the SoC. There's a tiny plug near the reset button where you can attach a 2032-type coin cell to maintain the time when the board is powered off, and there's library support. There's some example code in `.../libraries/OpenCR/examples/10. Etc/rtc_time`. The actual API is defined in `.../variants/OpenCR/hw/driver/drv_rtc.h`.

sunghowoo, 2025-02-24:
This issue has been inactive for a long time, I will proceed with closing it.  If you have any further concerns, please feel free to reopen the issue or create a new one.

closed 2025-02-24 by sunghowoo