ROBOTIS-GIT/OpenCR #318: JTAG/SWD Connector - How to use it?

Omar-Ismail, 2022-10-12:
How can the JTAG/SWD connector be used on the OpenCR board? What type of physical connection could be made to it (keeping in mind that the pins are 1.27mm apart which means normal Jumper wire can not be connected to it)?

I would like to program the OpenCR board using STM32CubeIDE is this possible? I'm not able to do so because of the need to use ST-Link V2 which can only be connected to the JTAG/SWD connector.

ROBOTIS-Will, 2022-10-13:
You can use .05" 10-pin connector like the one included in the [J-Link EDU Mini](https://www.segger.com/products/debug-probes/j-link/models/j-link-edu-mini/).
Please refer to the JTAG pinmap in the eManual below.
https://emanual.robotis.com/docs/en/parts/controller/opencr10/#layoutpin-map

Please be aware that burning a custom firmware and erasing the OpenCR bootloader will void product warranty.
Thank you.

Omar-Ismail, 2022-10-13:
Thank you for your reply and excuse me for my following long post. But maybe you can guide me because I'm still new to this. 

I wanted to program the OpenCR board using this Simulink package [https://www.mathworks.com/products/hardware/stm32.html] which is based on STM32CubeMX and in the settings the only connectivity mode is through ST-Link. That is why I was asking about how to use the JTAG/SWD connector.

If I'm not using the ST-Link and I'm just connecting the OpenCR board to my laptop through the USB, when I try to run or program the board I get an error that no STM32 MCU is detected.

So is it possible to know whether there is a workaround to connect the OpenCR board to Simulink?
And why did you assume that I might erase the OpenCR bootloader? Is it needed for such a connection? Does the bootloader offers some restricitions?

I was expecting that the board could be programmed using any programmer whether it is ArduinoIDE, STM32CubeIDE, or Simulink without the need of ST-Link or JTAG/SWD connection.

ROBOTIS-Will, 2022-10-13:
If you have correctly connected the JTAG/SWD pins to the ST-Link, but does not get detected from the software, make sure that the microcontroller of the OpenCR(STM32F746ZGT6) is supported in your development environment.

Uploading program using JTAG usually followed by erasing the entire memory and burn the chip which inevitably erases the bootloader.

Since it is an open source product, you may try using different development environment, but other development environments that are not written in the eManual such as STM32CubeIDE won't be supported from ROBOTIS.

Thank you.

Omar-Ismail, 2022-10-15:
Thank you for your reply. It helped a lot.

Right now, when I connect through the ST-Link V2, the device appears under Universal Serial Bus Devices. Is it possible to have it under the Ports (COM)? Is there a work around?

Without the ST-Link V2, it is possible to have a Virtual COM Port using the OpenCR default bootloader. However, once the ST-Link V2 is used to program the board the Virtual COM port disappears. So basically, I'm not able to have both ST-Link and Virtual COM port. Is there a solution for this?

Thank you.

Omar-Ismail, 2022-10-16:
Thank you for your reply. It helped a lot.

Can I know whether the USB connection allows serial communication and if yes on which USART?

Thank you.

closed 2022-10-21 by OI-93