ROBOTIS-GIT/OpenCR #321: Spurious readings from wheel position on Turtlebot3 via Dynamixel SDK

nigelgardiner, 2022-11-20:
I have tested this on several different Turtlebots. One I have updated to the latest firmware, the others may or may not be the latest, but the problem is consistent between them.

After travelling anywhere from 5-20 metres, both wheel positions suddenly jump to large values, and then jump back. This is ruinous for tracking odometry.

I've tracked the issue upstream to the Dynamixel SDK, which is outputting the large values, so the issue is either in the SDK or in the OpenCR board. I'm not sure how to pursue it further, but happy to investigate in more depth if anyone can give me pointers.

I've uploaded an image from plotjuggler showing the issue.

[image]

nigelgardiner, 2022-11-23:
I figure there are 3 places the problem could be... 
1. OpenCR code
2. Dynamixel motor firmware
3. Dynamixel SDK

Is that logical?

jobst-n, 2022-12-12:
Is this only happening with ros2 systems ?

closed 2025-11-03 by sunghowoo