ROBOTIS-GIT/OpenCR #334: Impossible to install 1.5.3 release on Arduino

Soupalognon, 2023-09-24:
When you try to install release 1.5.3 on Arduino by doing:

1. Go to Preferences --> add URL 
https://raw.githubusercontent.com/ROBOTIS-GIT/OpenCR/master/arduino/opencr_release/package_opencr_index.json

2. Go to Tools-->Board Manager
Install OpenCR 1.5.3

It never installs, error code : 
"Stream is not in the BZip2 format"

Roboteer1000, 2024-03-08:
For Windows 11, I have found a work around by copying the needed OpenCR folders and files into the Arduino OpenCR folder on C: drive (which is hidden and so you'll need Administrator privilege) which is listed below:

C:\Users\UserName\AppData\Local\Arduino15\packages\OpenCR\hardware\OpenCR\1.5.3  
(where UserName is your Windows Login name).

The folders and files, to be copied from, can be found in the OpenCR-1.5.3.zip file (downloaded from this GitHub), and they are in this folder:

D:\OpenCR-1.5.3.zip\OpenCR-1.5.3\arduino\opencr_arduino\opencr\

They are 4 folders named: "bootloaders", "cores", "libraries", and "variants"
and 5 files: ".gitignore", "boards.txt", "keywords.txt", "platform.txt" and "programmers.txt".

Once this copying is done, you can use Arduino IDE as "normal", I have checked this procedure with Arduino IDE 1.8.19 and 2.3.2.  However when you want to "Check Info" on this board, Arduino will say "Unknown Board", but it will list the VID and PID numbers.  Sketches for the OpenCR compiled, downloaded and executed fine on the OpenCR v.1.5.3 now.

sunghowoo, 2025-02-24:
This issue has been inactive for a long time, I will proceed with closing it.  If you have any further concerns, please feel free to reopen the issue or create a new one.

closed 2025-02-24 by sunghowoo