ROBOTIS-GIT/OpenCR #337: TB3/ROS2 OpenCR communication error

JinyongJeong-TIM, 2024-02-20:
I have checked closed issue related with this issue. 
https://github.com/ROBOTIS-GIT/OpenCR/issues/310

But I can't solve this issue with above guidline.

I am using Ardunino IDE 2.2.1, and installed Dynamixel2Arduino 0.6.0 & OpenCR board manager 1.5.0.

In the Arduino IDE, beep sound was not triggered after flash. As you know, normally, beep sound is triggered when binary file officially provided is flashed. 

And I met this error when I try to bring up turtlebot3

[turtlebot3_ros-3] [ERROR] [1708400526.385683042] [turtlebot3_node]: Failed connection with Devices

I have tried various combination of Dynamixel2Arduino & OpenCR board manager, but nothing had worked. 

Anyone can solve this issue?

shiba-8ro, 2024-02-23:
@JinyongJeong-TIM 
Will the second method to upload FW to OpenCR help you?
https://emanual.robotis.com/docs/en/platform/turtlebot3/opencr_setup/#opencr-setup-6
[image]

shiba-8ro, 2024-02-23:
@JinyongJeong-TIM 
You may have success if you try it on another PC, for example your friend's PC.
This may sound like a silly answer, but I've experienced something similar several times.

JinyongJeong-TIM, 2024-02-23:
@shiba-8ro So when you flash code with Arduino IDE with latest version code, Beep sound should be triggered, Right?

I tried to use other PC, but I didn't work.

shiba-8ro, 2024-02-23:
@JinyongJeong-TIM 
Thank you for your reply.
It turns out that the PC and Arduino IDE are fine.
In my experience, in cases like this, OpenCR 1.0 itself may be broken or malfunctioning (I don't know why, but sometimes OpenCR is in a bad mood even if it's not broken).
So, why not contact ROBOTIS CS?

sunghowoo, 2025-02-24:
Since this issue has been inactive for a long time, we will proceed with closing it. If you encounter any further problems, please feel free to reopen the issue or create a new one.

closed 2025-02-24 by sunghowoo