ROBOTIS-GIT/OpenCR #338: About default battery for OPENCR

minhlang102, 2024-03-14:
Hi guys,

I've read on manual that it said default battery is LI-PO 11.1V 1,800mAh 19.98Wh. I had a question that could i use 3s Lipo battery, 6200mah 100c for OpenCR via T-plug?

Thanks in advance,

shiba-8ro, 2024-03-16:
Hi @minhlang102 

OpenCR1.0 has a fuse.
I haven't tried high current LIPO, but you need to input power within the range that this fuse can safely use.
https://emanual.robotis.com/docs/en/parts/controller/opencr10/
[image]

minhlang102, 2024-03-16:
Hello @shiba-8ro 

Thank you for your response.
I'm using Turtlebot3 and it doesn't draw current more than 10A. So is it safety if i use battery mentioned above?

shiba-8ro, 2024-03-17:
@minhlang102 
I have never run a current of more than 5A through TB3, so I cannot guarantee it, but I think it is possible.
However, if you do so, please do so at your own risk.

langung102, 2024-03-17:
Thank you, @shiba-8ro
i'll consider it.

minhlang102, 2024-03-17:
Thank you, @shiba-8ro 
I'll consider it.

closed 2024-03-17 by minhlang102