ROBOTIS-GIT/OpenCR #339: Can't directly upload arduino code

qddxl, 2024-03-14:
When uploading a program via arduino, is it necessary to set opencr to bootmode every time? if not, the port cannot be found and cannot be uploaded successfully.

shiba-8ro, 2024-03-16:
Hi, @qddxl 
Did you press the button to enter DFU mode?
https://emanual.robotis.com/docs/en/parts/controller/opencr10/#enter-dfu-mode
[image]

sunghowoo, 2025-02-24:
Since this issue has been inactive for a long time, we will proceed with closing it. If you encounter any further problems, please feel free to reopen the issue or create a new one. Thank you for your understanding!

closed 2025-02-24 by sunghowoo